Lab-of-AI-and-Robotics/GS_ICP_SLAM #1: unable to install requirements

MikeCernea, 2024-03-22:
When trying to install requirements using 'pip install -r requirements.txt' most libs are not found with the required version(if at all i.e. actionlib) or require a different python version(i.e. contourpy-1.1.1).
Is it possible to review and update the requirements and ReadME to match?

Riboha, 2024-03-22:
Thank you for your attention to our work. I've updated both the requirements.txt file and the README. If you're still encountering the same issue, please let me know.

MikeCernea, 2024-03-25:
Hello Riboha.
Thank you for the update and fix on the requirements; all requirements were installed nicely.
For the submodules, however, it went poorly:
 - pip install submodules/diff-gaussian-rasterization fails with 
```
  Building wheel for simple-knn (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [120 lines of output]
...
UserWarning: The detected CUDA version (11.5) has a minor version mismatch with the version that was used to compile PyTorch (11.8). Most likely this shouldn't be a problem.
        warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
      building 'simple_knn._C' extension
...
/usr/include/c++/11/bits/std_function.h:435:145: error: parameter packs not expanded with ‘...’:
```
 - pip install submodules/simple-knn fails with the same error as above
 - submodules/fast_gicp cmake does not find PCL(I will work on installing that, is there a specific version or branch I should use?)

Kind regards,
Mike.

lbg030, 2024-03-28:
```
UserWarning: **The detected CUDA version (11.5) has a minor version mismatch with the version that was used to compile** PyTorch (11.8). Most likely this shouldn't be a problem.
        warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
```

As the error log shows, I think we should use the Cuda version at 11.8 or higher :)

closed 2024-04-06 by Riboha