Lab-of-AI-and-Robotics/GS_ICP_SLAM #2: Why not working?

lbg030, 2024-03-23:
I completely install requirements & submoules, then tried to run code.
But It is not working.
I think backend process is deadlocked, What should I do?\

[image]

Riboha, 2024-03-23:
The problem seems to have occurred because the system was started before the mapping process was fully prepared.
The tracking process reads the dataset, while mapping process is implemented to wait until the dataset is fully loaded. So this issue was unexpected and had not occurred in my environment before.
Thank you for raising this issue, and I have updated the code to address it. 
Please let me know if the same issue occurs again.

hjr37, 2024-03-28:
Is the camera tracking performed in an already established or nearly established 3D Gaussian Field? I think that even the traditional ICP method struggles to achieve decent rendering quality while performing tracking at a speed of 100fps.

kinghangwong, 2024-03-30:
Hi, I have just cloned the repo but am still experiencing the same error as @lbg030. Running on an RTX 4090.

> The problem seems to have occurred because the system was started before the mapping process was fully prepared. The tracking process reads the dataset, while mapping process is implemented to wait [...]

Riboha, 2024-03-30:
@ztmo520 Thank you for your efforts and for sharing your experience!!!
For GS-ICP SLAM, we use a modified version of our fast_gicp, which is on gs_icp_slam branch of it.
The python_tester is for the main branch of fast_gicp, and does not work with the gs_icp_slam version.

I'm testing GS-ICP SLAM in other environments, trying to reproduce this error. However, I haven't encountered this problem yet. Can you share some information about the environment of RTX 3060 desktop? (version of Ubuntu, cuda, torch ...) It will be helpful for me to tackle this problem!!

> Thank you for your amazing work!
> 
> I experimented on two machines and on one of them, an RTX 3090, everything went fine, on the other, an RTX 3060, the same problem would occur.
> 
> I found that the code would get stuck at the location in Tracker, self.reg.align().
> 
> https://github.com/Lab-of-AI-and-Robotics/GS_ICP_SLAM/blob/53fd4e60091ab2a099d817eba1abb349d6d3481a/mp_Tracker.py#L169
> 
> I did the python_tester test as instructed in the submodule /fast_gicp, and strangely, on both my computers, it gives me an error when running into the "align" position, segmentation fault (core dumped) or MemoryError: std::bad_alloc. This makes me very confused, because in gs_icp_slam.py it works fine on RTX 3090 machine.
> 
> I hope this info will give you some clues.

ztmo520, 2024-03-31:
@Riboha I experimented again on a new RTX 4090 machine and it is working fine. I found that it is fine on an ubuntu 20.04 machine, and will have problems on 22.04, I'm guessing it's an environment issue with compiling fast_gicp that is preventing it from working properly，like g++, pcl, vtk, etc.

Riboha, 2024-03-31:
Thanks a lot @ztmo520!!!
I will test our system on ubuntu 22.04. If I can't find a solution to this problem, I'll provide a Docker image.

while0l1, 2024-04-18:
For my ubuntu 20.04, downgrading pcl from version 1.12 to 1.10 solved the problem.

Buffyqsf, 2024-04-24:
> Thanks a lot @ztmo520!!! I will test our system on ubuntu 22.04. If I can't find a solution to this problem, I'll provide a Docker image.

Hi have your tried on ubuntu22.04? I think I encountered the same problem.

BetterYimi, 2024-05-09:
> For my ubuntu 20.04, downgrading pcl from version 1.12 to 1.10 solved the problem.

Hi, My pcl version also is 1.10 but fast_gicp still can not work,.Could you please tell me what  your openmp version and vtk version are？

closed 2024-07-06 by Riboha