Lab-of-AI-and-Robotics/GS_ICP_SLAM #4: Visualization Error: Problem loading model info from input path dataset/TUM/rgbd_dataset_freiburg1_desk

LeonGoretzkatju, 2024-03-30:
Hello,

Thank you so much for your amazing work! When I'm trying to run your code with verbose=True, and I also installed the SIBR_viewers on Ubuntu20.04, then I start the `./SIBR_remoteGaussian_app` command, and I run the gs_icp_slam code, but after connecting to the server, the main GUI crashed and report the folllowing error:
`[SIBR] --  INFOS  --:	Initialization of GLFW
[SIBR] --  INFOS  --:	OpenGL Version: 4.6.0 NVIDIA 525.147.05[major: 4, minor: 6]
[SIBR] --  INFOS  --:	Trying to connect...
[SIBR] --  INFOS  --:	Connected!
[SIBR] ##  ERROR  ##:	FILE /home/alex/research/GS_ICP_SLAM/SIBR_viewers/src/core/scene/ParseData.cpp
			LINE 560, FUNC getParsedData
			Cannot determine type of dataset at /dataset/TUM/rgbd_dataset_freiburg1_desk
[SIBR] ##  ERROR  ##:	FILE /home/alex/research/GS_ICP_SLAM/SIBR_viewers/src/projects/remote/apps/remoteGaussianUI/main.cpp
			LINE 54, FUNC resetScene
			Problem loading model info from input path dataset/TUM/rgbd_dataset_freiburg1_desk. Consider overriding path to model directory using --path.terminate called after throwing an instance of 'std::runtime_error'
  what():  See log for message errors
Aborted (core dumped)`
I was wondering if you have met with the same problem or if you can provided some suggestions? Thank you!

Riboha, 2024-03-31:
Thank you for reporting this problem!
It was my mistake, and I updated SIBR viewer.
Is it working now?

LeonGoretzkatju, 2024-03-31:
> Thank you for reporting this problem! It was my mistake, and I updated SIBR viewer. Is it working now?

Thank you so much for your help, I have successfully open the SIBR remote viewer, but after I open the remote window and run the tum.sh, I found there is no point in the visualization window, like the following image,
[image]
The only different point is that, your SIBR viewer need opencv version 4.5, but because my ubuntu is 20.04, so I change it to the OpenCV 4.2, I'm not sure if this version difference cause my problem, have you encountered with the similar issue in the figure? or could you please tell me the right procedure to remote visualize the SLAM result real-time? Thank you so much!

LeonGoretzkatju, 2024-03-31:
> Thank you for reporting this problem! It was my mistake, and I updated SIBR viewer. Is it working now?

Also the remote terminal has the following warning,
[image]
I guess this maybe caused by lack of .bin 3D file

Riboha, 2024-03-31:
I haven't aligned the initial viewer pose with the reconstructed 3D world yet, so you need to move manually to see the reconstructed side. 
I apologize for the inconvenience. This repository isn't perfect in terms of user-friendliness, but I'll continuously update it.

RoseWillow, 2024-04-04:
> > Thank you for reporting this problem! It was my mistake, and I updated SIBR viewer. Is it working now?
> 
> Also the remote terminal has the following warning, ![Screenshot from 2024-03-31 20-04-13](https://private-user-images.githubusercontent.com/53984646/318260314-59d23f65-2e21-420b-8134-77ea54ffc5d2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIyNDgxMTYsIm5iZiI6MTcxMjI0NzgxNiwicGF0aCI6Ii81Mzk4NDY0Ni8zMTgyNjAzMTQtNTlkMjNmNjUtMmUyMS00MjBiLTgxMzQtNzdlYTU0ZmZjNWQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA0VDE2MjMzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5YjQ5ZGY5MTBhYjc4M2U1YTVlNTlkMjgxZjYzYWM1ZDY2NDNiMjdkZjhlNzcwMTk4Y2FjYWM0MjQ0MTVhODImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.E5FMDQPXNfHpImXi7R0k9kX3GUCAukIxtHeOyuKHHLA) I guess this maybe caused by lack of .bin 3D file

Hi, I got the same error, have you solved it ?

LeonGoretzkatju, 2024-04-09:
> Hi, I got the same error, have you solved it ?

I haven't solved yet, I'm trying to find some relative code about how to align net image with initial coordinate of SIBR viewer

closed 2024-05-09 by Riboha