Lab-of-AI-and-Robotics/GS_ICP_SLAM #6: Nothing in /results after running

JinXJinLeX, 2024-04-03:
There is nothing in ../experiments/results/ after I run tum.sh, is it the only way to visualize by SIBR_viewer? How can I get a traj or .ply file?

LemonSoda-RPG, 2024-04-08:
I encountered the same issue while running it with Docker.
There is nothing in experiments/results

youzi260, 2024-04-09:
I also encountered the same problem. There is nothing under the result. Have you solved this problem?

Riboha, 2024-04-09:
Exporting results are not implemented, but I'm working on it!
It will be added soon.

JinXJinLeX, 2024-04-10:
> Exporting results are not implemented, but I'm working on it! It will be added soon.

Looking forward to it! Thanks for your excellent work!

KwanWaiPang, 2024-04-11:
> There is nothing in ../experiments/results/ after I run tum.sh, is it the only way to visualize by SIBR_viewer? How can I get a traj or .ply file?

hi~do you successfully use the SIBR_viewer to visualize the result? I can not get any result after I run the ./install/bin/SIBR_remoteGaussian_app --rendering-size 1280 720

[image]

Saligz, 2024-06-12:
Hi, did you launch the slam algorithm before the visualization tool ?

closed 2024-07-06 by Riboha