Lab-of-AI-and-Robotics/GS_ICP_SLAM #10: Visualization Error: X11 Failed in Docker

shuttworth, 2024-04-07:
when run  `./install/bin/SIBR_remoteGaussian_app` 
[SIBR] -- INFOS --:	Initialization of GLFW
[SIBR] ## ERROR ##:	FILE /home/SIBR_viewers/src/core/graphics/Window.cpp
			LINE 30, FUNC glfwErrorCallback
			X11: Failed to open display localhost:12.0
terminate called after throwing an instance of 'std::runtime_error'
 what(): See log for message errors
Aborted (core dumped)

How can i achieve visualization in docker?

Riboha, 2024-04-08:
Type "xhost +" in the terminal of main environment, and then try viewer again.

shuttworth, 2024-04-08:
hello, "xhost +" in the terminal of main environment doesn't work, still "X11: Failed to open display localhost:12.0". 
Many methods have been tried, but none can work.
Does any one successfully achieve visualization in the docker?

shuttworth, 2024-04-08:
additional, when run `python -W ignore gs_icp_slam.py --dataset_path dataset/Replica/office1 --verbose`
100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2000/2000 [00:53<00:00, 37.32it/s]
Waiting for mapping process to be prepared
  0%|                                                                                                                                               | 0/2000 [00:00<?, ?it/s]qt.qpa.xcb: could not connect to display localhost:11.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/usr/local/lib/python3.9/dist-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

How can i achieve visualization in the docker?

Eku127, 2024-04-09:
Try docker run with `-e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix` commends.

If it doesn't work, execute `export MESA_GL_VERSION_OVERRIDE=4.5` within the container.

shuttworth, 2024-04-09:
[image]
Unfortunately, this didn't work; there were still issues with --verbose visualization within docker, even after executing `export MESA_GL_VERSION_OVERRIDE=4.5` ,and executing `docker run -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY xxxxxx` in the beginning. System is Ubuntu22.04, cuda 11.8

674123, 2024-12-17:
> > 尝试在容器内执行。`export MESA_GL_VERSION_OVERRIDE=4.5`
> 
> ![图像](https://private-user-images.githubusercontent.com/46838923/320774443-4d6427a2-9d4f-4785-a16d-90e125cf48b1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ0MjY4MzUsIm5iZiI6MTczNDQyNjUzNSwicGF0aCI6Ii80NjgzODkyMy8zMjA3NzQ0NDMtNGQ2NDI3YTItOWQ0Zi00Nzg1LWExNmQtOTBlMTI1Y2Y0OGIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE3VDA5MDg1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5ZDU0YmNlYTdlODgwNzcxYTY5MjBmNmM3MWZjODRlNGE0NjAwMjczYjg1OTgzMzZkZTc2NTM3MDhiZTE5NmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.tadHJzdNGZcFROZuDUk6v0JLTaBb7hgQZ-xDvK_OGo8)不幸的是，这没有奏效;Docker 中的 --verbose 可视化仍然存在问题，即使在执行 ，并且一开始执行之后也是如此。系统为 Ubuntu22.04， cuda 11.8`export MESA_GL_VERSION_OVERRIDE=4.5``docker run -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY xxxxxx`
I have the same problem. Have you solved it?

674123, 2024-12-17:
> 运行时 [SIBR] -- INFOS --： 初始化 GLFW [SIBR] ## 错误 ##： 文件 /home/SIBR_viewers/src/core/graphics/Window.cpp 第 30 行，FUNC glfwErrorCallback X11： 在抛出 'std：：runtime_error' 的实例后，无法打开显示 localhost：12.0 终止调用 what（） 的实例：请参阅消息错误日志 中止（核心转储）`./install/bin/SIBR_remoteGaussian_app`
> 
> 如何在 docker 中实现可视化？

I have the same problem. Have you solved it?

LHX1220328779, 2025-04-18:
**Before creating a container, run the following program:**
`XSOCK=/tmp/.X11-unix && XAUTH=~/.docker.xauth && touch $XAUTH && xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge - && xhost +`
**When creating a container later, add:**
`-v $XSOCK:$XSOCK -v $XAUTH:/root/.Xauthority -e DISPLAY=$DISPLAY -e XAUTHORITY=/root/.Xauthority`
**If it malfunctions afterwards, execute it again outside the container:**
`XSOCK=/tmp/.X11-unix && XAUTH=~/.docker.xauth && touch $XAUTH && xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge - && xhost +`
**Restart the container again.**
**Ensure that the display information of `echo $DISPLAY` on the outside and inside of the container is the same.**