Lab-of-AI-and-Robotics/GS_ICP_SLAM #13: TypeError: cannot pickle 'pygicp.FastGICP' object

KwanWaiPang, 2024-04-11:
Dear author,
First of all, I would like to thank the open source code for your excellent work!
I completely install requirements & submoules, then tried to run code.
But It doesn't work.

[image]

  File "/home/gwp/miniconda3/envs/gsicpslam/lib/python3.9/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: cannot pickle 'pygicp.FastGICP' object
[W CudaIPCTypes.cpp:15] Producer process has been terminated before all shared CUDA tensors released. See Note [Sharing CUDA tensors]

KwanWaiPang, 2024-04-11:
This problem has be solved by install the PCL correctly:
pip install pcl
conda config --add channels conda-forge
conda install -c sirokujira python-pcl

closed 2024-04-11 by KwanWaiPang

Aeson-Hsu, 2024-04-16:
I have the same problem, and I follow your solution install pcl and python-pcl. 
But this problem still remains.
Do you have some advice to solve this?

pip list  as the follow:
``` 
Package                       Version
----------------------------- -----------
actionlib                     1.12.1
addict                        2.4.0
angles                        1.9.12
attrs                         23.2.0
backcall                      0.2.0
bondpy                        1.8.5
brotlipy                      0.7.0
camera_calibration            1.15.2
camera_calibration_parsers    1.11.13
catkin                        0.7.29
certifi                       2022.12.7
cffi                          1.15.0
charset-normalizer            2.0.4
click                         8.1.7
comm                          0.1.4
ConfigArgParse                1.7
controller_manager            0.18.4
controller_manager_msgs       0.18.4
cryptography                  39.0.1
cv_bridge                     1.13.1
cycler                        0.11.0
Cython                        3.0.10
dash                          2.15.0
dash-core-components          2.0.0
dash-html-components          2.0.0
dash-table                    5.0.0
decorator                     5.1.1
diagnostic_analysis           1.9.7
diagnostic_common_diagnostics 1.9.7
diagnostic_updater            1.9.7
diff-gaussian-rasterization   0.0.0
dynamic_reconfigure           1.6.5
fastjsonschema                2.19.1
filelock                      3.12.2
Flask                         2.2.5
flit_core                     3.6.0
fonttools                     4.38.0
gazebo_plugins                2.8.7
gazebo_ros                    2.8.7
gencpp                        0.6.6
geneus                        2.2.6
genlisp                       0.4.16
genmsg                        0.5.17
gennodejs                     2.0.1
genpy                         0.6.16
idna                          3.4
image_geometry                1.13.1
importlib-metadata            6.7.0
importlib-resources           5.12.0
interactive-markers           1.11.5
ipython                       7.34.0
ipywidgets                    8.1.2
itsdangerous                  2.1.2
jedi                          0.19.1
Jinja2                        3.1.3
joblib                        1.3.2
joint_state_publisher         1.12.15
jsonschema                    4.17.3
jupyter_core                  4.12.0
jupyterlab_widgets            3.0.10
kdl-parser-py                 1.13.3
kiwisolver                    1.4.5
[... 110 lines cut ...]
tf                            1.12.1
tf_conversions                1.12.1
tf2_geometry_msgs             0.6.5
tf2_kdl                       0.6.5
tf2_py                        0.6.5
tf2_ros                       0.6.5
threadpoolctl                 3.1.0
topic_tools                   1.14.13
torch                         1.12.1
torchaudio                    0.12.1
torchmetrics                  0.11.4
torchvision                   0.13.1
tqdm                          4.66.2
traitlets                     5.9.0
typing_extensions             4.4.0
urdfdom-py                    0.4.6
urllib3                       1.26.14
wcwidth                       0.2.13
Werkzeug                      2.2.3
wheel                         0.38.4
widgetsnbextension            4.0.10
xacro                         1.13.20
zipp                          3.15.0

```

kk6398, 2024-04-16:
> I have the same problem, and I follow your solution install pcl and python-pcl. But this problem still remains. Do you have some advice to solve this?
>
> pip list as the follow:
>
> ``` [...]

The same issue I met with you.

kk6398, 2024-04-16:
Do you have any methods to solve it??? Thank you

KwanWaiPang, 2024-04-16:
I write a blog for the setup and testing of the GS-ICP-SLAM.
Hopefully, this can help to solve the problem.
And my suggestion is: use the conda to setup the environment following the guidance. And install the PCL using the follwoing command
`
pip install pcl
conda config --add channels conda-forge
conda install -c sirokujira python-pcl   
`
[CSDN blog](https://blog.csdn.net/gwplovekimi/article/details/137587778?spm=1001.2014.3001.5501)

scut666, 2024-05-13:
I meet the same problem and I follow your solution install pcl and python-pcl. But this problem still remains. Do you have some advice to solve this?

pip list as follow:

action-msgs                          1.2.1
action-tutorials-interfaces          0.20.3
action-tutorials-py                  0.20.3
actionlib-msgs                       4.2.3
addict                               2.4.0
ament-cmake-test                     1.3.8
ament-copyright                      0.12.10
ament-cppcheck                       0.12.10
ament-cpplint                        0.12.10
ament-flake8                         0.12.10
ament-index-python                   1.4.0
ament-lint                           0.12.10
ament-lint-cmake                     0.12.10
ament-package                        0.14.0
ament-pep257                         0.12.10
ament-uncrustify                     0.12.10
ament-xmllint                        0.12.10
angles                               1.15.0
asttokens                            2.4.1
attrs                                23.2.0
blinker                              1.8.2
Brotli                               1.0.9
builtin-interfaces                   1.2.1
catkin-pkg                           1.0.0
certifi                              2024.2.2
charset-normalizer                   3.3.2
click                                8.1.7
colorama                             0.4.6
comm                                 0.2.2
composition-interfaces               1.2.1
ConfigArgParse                       1.7
contourpy                            1.2.1
cv-bridge                            3.2.1
cycler                               0.12.1
dash                                 2.17.0
dash-core-components                 2.0.0
dash-html-components                 2.0.0
dash-table                           5.0.0
decorator                            5.1.1
demo-nodes-py                        0.20.3
diagnostic-msgs                      4.2.3
diff_gaussian_rasterization          0.0.0
docutils                             0.21.2
domain-coordinator                   0.10.0
example-interfaces                   0.9.3
examples-rclpy-executors             0.15.1
examples-rclpy-minimal-action-client 0.15.1
examples-rclpy-minimal-action-server 0.15.1
examples-rclpy-minimal-client        0.15.1
examples-rclpy-minimal-publisher     0.15.1
examples-rclpy-minimal-service       0.15.1
examples-rclpy-minimal-subscriber    0.15.1
exceptiongroup                       1.2.1
executing                            2.0.1
fastjsonschema                       2.19.1
Flask                                3.0.3
fonttools                            4.51.0
geometry-msgs                        4.2.3
idna                                 3.7
image-geometry                       3.2.1
importlib_metadata                   7.1.0
interactive-markers                  2.3.2
ipython                              8.24.0
ipywidgets                           8.1.2
itsdangerous                         2.2.0
jedi                                 0.19.1
[... 140 lines cut ...]
tf2-kdl                              0.25.6
tf2-msgs                             0.25.6
tf2-py                               0.25.6
tf2-ros-py                           0.25.6
tf2-tools                            0.25.6
threadpoolctl                        3.5.0
topic-monitor                        0.20.3
torch                                1.12.1
torchaudio                           0.12.1
torchmetrics                         1.4.0
torchvision                          0.13.1
tqdm                                 4.66.4
traitlets                            5.14.3
trajectory-msgs                      4.2.3
turtlesim                            1.4.2
typing_extensions                    4.11.0
tzdata                               2024.1
unique-identifier-msgs               2.2.1
urllib3                              2.2.1
visualization-msgs                   4.2.3
wcwidth                              0.2.13
Werkzeug                             3.0.3
wheel                                0.43.0
widgetsnbextension                   4.0.10
zipp                                 3.18.1

huarzail, 2024-07-17:
I had encountered this same error. You can attempt to change the fast-gicp branch of gs_fast_slm.

scpoppin1207, 2024-10-26:
I've tried this method but error still occurs.
Then I changed the fast-gicp branch of gs_fast_slm but it doesn't work, neither.
HOW TO SOLVE THIS ISSUE?

> I write a blog for the setup and testing of the GS-ICP-SLAM. Hopefully, this can help to solve the problem. And my suggestion is: use the conda to setup the environment following the guidance. And [...]