Lab-of-AI-and-Robotics/GS_ICP_SLAM #14: cd SIBR_viewers install problem.

Dreambyland, 2024-04-12:
How do you install/build real-team view ?

I've tried with command cmake$ make , but faced the problems:
[ 89%] Building CUDA object ../extlibs/CudaRasterizer/build/CMakeFiles/CudaRasterizer.dir/cuda_rasterizer/backward.cu.o
nvcc fatal   : Unknown option 'fpermissive'
make[2]: *** [../extlibs/CudaRasterizer/build/CMakeFiles/CudaRasterizer.dir/build.make:76: ../extlibs/CudaRasterizer/build/CMakeFiles/CudaRasterizer.dir/cuda_rasterizer/backward.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1473: ../extlibs/CudaRasterizer/build/CMakeFiles/CudaRasterizer.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

JinXJinLeX, 2024-04-12:
Hi, I met the same problem, solved it by updating my cmake version to 3.22.3

Dreambyland, 2024-04-15:
> Hi, I met the same problem, solved it by updating my cmake version to 3.22.3

it didn't help me. still facing the problem

closed 2024-05-10 by Dreambyland