Lab-of-AI-and-Robotics/GS_ICP_SLAM #15: cmake --build build -j24 --target install error

Tanfuchy, 2024-04-13:
Viewer
cd SIBR_viewers
cmake -Bbuild . -DCMAKE_BUILD_TYPE=Release
cmake --build build -j24 --target install

After entering this, how to solve the content that appears?

cmake --build build -j24 --target install

[ 91%] Linking CXX executable SIBR_texturedMesh_app
[ 91%] Linking CXX executable SIBR_PointBased_app
[ 93%] Built target sibr_remote
[ 95%] Built target sibr_gaussian
[ 95%] Linking CXX executable SIBR_remoteGaussian_app
[ 95%] Linking CXX executable SIBR_gaussianViewer_app
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::optflow::calcOpticalFlowSF(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, int, int)'
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::ximgproc::guidedFilter(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, double, int)'
/usr/bin/ld: /home/oem/anaconda3/lib/libgio-2.0.so: undefined reference to `g_module_open_full'
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::optflow::createOptFlow_DeepFlow()'
collect2: error: ld returned 1 exit status
make[2]: *** [src/projects/basic/apps/texturedMesh/CMakeFiles/SIBR_texturedMesh_app.dir/build.make:151: src/projects/basic/apps/texturedMesh/SIBR_texturedMesh_app] Error 1
make[1]: *** [CMakeFiles/Makefile2:1335: src/projects/basic/apps/texturedMesh/CMakeFiles/SIBR_texturedMesh_app.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::optflow::calcOpticalFlowSF(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, int, int)'
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::ximgproc::guidedFilter(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, double, int)'
/usr/bin/ld: /home/oem/anaconda3/lib/libgio-2.0.so: undefined reference to `g_module_open_full'
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::optflow::createOptFlow_DeepFlow()'
collect2: error: ld returned 1 exit status
make[2]: *** [src/projects/basic/apps/pointBased/CMakeFiles/SIBR_PointBased_app.dir/build.make:151: src/projects/basic/apps/pointBased/SIBR_PointBased_app] Error 1
make[1]: *** [CMakeFiles/Makefile2:1424: src/projects/basic/apps/pointBased/CMakeFiles/SIBR_PointBased_app.dir/all] Error 2
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::optflow::calcOpticalFlowSF(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, int, int)'
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::ximgproc::guidedFilter(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, double, int)'
/usr/bin/ld: /home/oem/anaconda3/lib/libgio-2.0.so: undefined reference to `g_module_open_full'
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::optflow::createOptFlow_DeepFlow()'
collect2: error: ld returned 1 exit status
make[2]: *** [src/projects/remote/apps/remoteGaussianUI/CMakeFiles/SIBR_remoteGaussian_app.dir/build.make:152: src/projects/remote/apps/remoteGaussianUI/SIBR_remoteGaussian_app] Error 1
make[1]: *** [CMakeFiles/Makefile2:1871: src/projects/remote/apps/remoteGaussianUI/CMakeFiles/SIBR_remoteGaussian_app.dir/all] Error 2
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::optflow::calcOpticalFlowSF(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, int, int)'
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::ximgproc::guidedFilter(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, double, int)'
/usr/bin/ld: /home/oem/anaconda3/lib/libgio-2.0.so: undefined reference to `g_module_open_full'
/usr/bin/ld: ../../../../core/video/libsibr_video.so: undefined reference to `cv::optflow::createOptFlow_DeepFlow()'
collect2: error: ld returned 1 exit status
make[2]: *** [src/projects/gaussianviewer/apps/gaussianViewer/CMakeFiles/SIBR_gaussianViewer_app.dir/build.make:153: src/projects/gaussianviewer/apps/gaussianViewer/SIBR_gaussianViewer_app] Error 1
make[1]: *** [CMakeFiles/Makefile2:1609: src/projects/gaussianviewer/apps/gaussianViewer/CMakeFiles/SIBR_gaussianViewer_app.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

Riboha, 2024-04-15:
Hello,

If I remember correctly, optflow is included in opencv-contrib library, not the original opencv.
Maybe you need to install opencv-contrib library.

Tanfuchy, 2024-04-16:
thanks，but I have installed the opencv-contrib library.

Zqzne, 2024-11-07:
I have encountered the same problem, did you solve it?