Lab-of-AI-and-Robotics/GS_ICP_SLAM #18: Sparse reconstruction test

Shanzhaguoo, 2024-04-23:
Hello! I have a question I'd like to ask: The original Replica dataset had 2,000 input images, and now I have performed some sparsification (either 800 or 400 images). How should I adjust the original parameters? 
--keyframe_th 0.7
--knn_maxd 99999.0
--overlapped_th 5e-4
--max_correspondence_distance 0.02
--trackable_opacity_th 0.05
--overlapped_th2 5e-5
--downsample_rate 10
Maybe it's not very good for sparse reconstruction, but I just want to see the effects and conduct a trial, could you give me some suggestions? Thanks!

Riboha, 2024-04-23:
Hello!

Unfortunately, it is not implemented in our system.
I think you can manually choose sparse input images among the original 2,000 images, and remove other images.
And you need to modify the traj.txt.

Shanzhaguoo, 2024-04-23:
> Hello!
> 
> Unfortunately, it is not implemented in our system. I think you can manually choose sparse input images among the original 2,000 images, and remove other images. And you need to modify the traj.txt.

Thanks for your advice! I've tried this idea on Replica, and the psnr dropped to 20, approximately.
My parameter settings are as follows:
    --keyframe_th 0.1 \
    --knn_maxd 99999.0 \
    --overlapped_th 1e-4 \
    --max_correspondence_distance 0.02 \
    --trackable_opacity_th 0.05 \
    --overlapped_th2 1e-5 \
    --downsample_rate 2 \
 It's possible that the parameters I've set are not quite appropriate.

closed 2024-04-25 by Riboha