Lab-of-AI-and-Robotics/GS_ICP_SLAM #20: About visualizing results

youzi260, 2024-04-24:
Thanks for the update on code saving results.
First of all, the SIBR_viewer you provided cannot be compiled and is stuck. I think other people have also encountered this problem. The sibr_viewer of 3dgs does not support the ply file you output. Can you provide a SIBR_viewer that can be compiled?
So I chose rerun.io to visualize the TUM training results. The error message showed that the traj.txt file was missing. This file is in the Replica data set. Is this visualization code only adapted to replica?

Riboha, 2024-04-24:
Hello!

As you mentioned, some users are suffering from problems in SIBR_viewer.
Most of the problems in compiling SIBR_viewer are caused by environmental differences, and it is not easy to help...
That is why I'm trying to make viewers with rerun.io, and currently, I have no plan to update SIBR_viewer.
And I'm planning to add a viewer with rerun.io which can load the .ply file and visualize it.

Currently implemented rerun.io viewer is for both replica and TUM datasets and worked well when I tested.
Maybe the error you got is caused by another reason. 
Does the system work properly without rerun.io viewer but goes bad when using rerun.io viewer? Can you explain a little more about your situation?

youzi260, 2024-04-25:
Thank you for your reply. Before I run "python -W ignore gs_icp_slam.py --rerun_viewer", I modified the dataset_path in gs_icp_slam.py to be the absolute path where the TUM dataset is located, and the output_path is ./experments/results after training the TUM dataset. In this result Many  .
[image]
png pictures and scene.ply files are saved. The error is reported as follows:

Riboha, 2024-05-02:
Sorry for the late response.
You need to change the directory of caminfo file! Like this,
python gs_icp_slam.py --[other arguments...] --config configs/TUM/[scene].txt

closed 2024-05-09 by Riboha