Lab-of-AI-and-Robotics/GS_ICP_SLAM #25: fastgicp error

EH-HU, 2024-05-09:
AttributeError:
'pygicp.FastGICP' object has no attribute 'set_max_knn_distance'

How can I solve this problem?

EH-HU, 2024-05-09:
Add. This code will produce a corresponding error
self.reg.set_input_target(points)

Segmentation fault (core dumped)

Riboha, 2024-05-09:
Did you clone and install the fast_gicp on the gs_icp_slam branch? (Not on the main branch)

EH-HU, 2024-05-09:
Thank you for your response, the first error has been fixed, but the second error is still not fixed.

huarzail, 2024-07-17:
How to solve this problem??? I also have this problem.
I found that the official code of fase_gicp used the main branch. But when I run the command "bash tum.sh", it was not well running. It endless loop at the second iter(current_pose = self.reg.align(initial_pose)).
When I change the fase_gicp brance of gs_gicp_slam, the error changed as : AttributeError: 'pygicp.FastGICP' object has no attribute 'set_max_knn_distance'. What should I do???

When I use the gdb debug, I found the endless loop is this log:
===============================================================
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fffc3b0221c in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > > >::calculate_source_covariances_with_filter<pcl::PointXYZ>(pcl::PointCloud<pcl::PointXYZ>::ConstPtr const&, pcl::search::Search<pcl::PointXYZ>&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator<float> >&, std::vector<float, std::allocator<float> >&, std::vector<int, std::allocator<int> >&) [clone ._omp_fn.0] () from /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
(gdb) backtrace
#0  0x00007fffc3b0221c in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > > >::calculate_source_covariances_with_filter<pcl::PointXYZ>(pcl::PointCloud<pcl::PointXYZ>::ConstPtr const&, pcl::search::Search<pcl::PointXYZ>&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator<float> >&, std::vector<float, std::allocator<float> >&, std::vector<int, std::allocator<int> >&) [clone ._omp_fn.0] ()
    at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
#1  0x00007fffc39ae686 in GOMP_parallel () at /lib/x86_64-linux-gnu/libgomp.so.1
#2  0x00007fffc3b0f226 in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > > >::calculate_source_covariances_with_filter<pcl::PointXYZ>(pcl::PointCloud<pcl::PointXYZ>::ConstPtr const&, pcl::search::Search<pcl::PointXYZ>&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator<float> >&, std::vector<float, std::allocator<float> >&, std::vector<int, std::allocator<int> >&) ()
    at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
#3  0x00007fffc3b0f4c1 in fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > > >::computeTransformation(pcl::PointCloud<pcl::PointXYZ>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&) ()
    at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
#4  0x00007fffc3b5b25e in  () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
#5  0x00007fffc3b5bcef in  () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
#6  0x00007fffc3b4f2ac in  () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
#7  0x00005555556b3e0e in  ()
#8  0x00005555556aa5eb in _PyObject_MakeTpCall ()
#9  0x00005555556c27bb in  ()
#10 0x00005555556a28a2 in _PyEval_EvalFrameDefault ()
#11 0x00005555556b470c in _PyFunction_Vectorcall ()
#12 0x000055555569ce0d in _PyEval_EvalFrameDefault ()
#13 0x000055555578de56 in  ()
#14 0x000055555578dcf6 in PyEval_EvalCode ()
#15 0x00005555557b87d8 in  ()
#16 0x00005555557b20bb in  ()
#17 0x00005555557b8525 in  ()
#18 0x00005555557b7a08 in _PyRun_SimpleFileObject ()
#19 0x00005555557b7653 in _PyRun_AnyFileObject ()
#20 0x00005555557aa41e in Py_RunMain ()
#21 0x0000555555780cad in Py_BytesMain ()
#22 0x00007ffff7c29d90 in __libc_start_call_main (main=main@entry=0x555555780c70, argc=argc@entry=6, argv=argv@entry=0x7fffffffdfb8) at ../sysdeps/nptl/libc_start_call_main.h:58
#23 0x00007ffff7c29e40 in __libc_start_main_impl
     (main=0x555555780c70, argc=6, argv=0x7fffffffdfb8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdfa8) at ../csu/libc-start.c:392
#24 0x0000555555780ba5 in _start ()
============================================================================

huarzail, 2024-07-17:
How to solve this problem??? I also have this problem.
I found that the official code of fase_gicp used the main branch. But when I run the command "bash tum.sh", it was not well running. It endless loop at the second iter(current_pose = self.reg.align(initial_pose)).
When I change the fase_gicp brance of gs_gicp_slam, the error changed as : AttributeError: 'pygicp.FastGICP' object has no attribute 'set_max_knn_distance'. What should I do???

When I use the gdb debug, I found the endless loop is this log:
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fffc3b0221c in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > > >::calculate_source_covariances_with_filterpcl::PointXYZ(pcl::PointCloudpcl::PointXYZ::ConstPtr const&, pcl::search::Searchpcl::PointXYZ&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator >&, std::vector<float, std::allocator >&, std::vector<int, std::allocator >&) [clone ._omp_fn.0] () from /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
(gdb) backtrace
#0 0x00007fffc3b0221c in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > > >::calculate_source_covariances_with_filterpcl::PointXYZ(pcl::PointCloudpcl::PointXYZ::ConstPtr const&, pcl::search::Searchpcl::PointXYZ&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator >&, std::vector<float, std::allocator >&, std::vector<int, std::allocator >&) [clone ._omp_fn.0] ()
at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
#1 0x00007fffc39ae686 in GOMP_parallel () at /lib/x86_64-linux-gnu/libgomp.so.1
#2 0x00007fffc3b0f226 in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > > >::calculate_source_covariances_with_filterpcl::PointXYZ(pcl::PointCloudpcl::PointXYZ::ConstPtr const&, pcl::search::Searchpcl::PointXYZ&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator >&, std::vector<float, std::allocator >&, std::vector<int, std::allocator >&) ()
at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
#3 0x00007fffc3b0f4c1 in fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > > >::computeTransformation(pcl::PointCloudpcl::PointXYZ&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&) ()
at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
#4 0x00007fffc3b5b25e in () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
#5 0x00007fffc3b5bcef in () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
#6 0x00007fffc3b4f2ac in () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
#7 0x00005555556b3e0e in ()
#8 0x00005555556aa5eb in _PyObject_MakeTpCall ()
#9 0x00005555556c27bb in ()
#10 0x00005555556a28a2 in _PyEval_EvalFrameDefault ()
#11 0x00005555556b470c in _PyFunction_Vectorcall ()
#12 0x000055555569ce0d in _PyEval_EvalFrameDefault ()
#13 0x000055555578de56 in ()
#14 0x000055555578dcf6 in PyEval_EvalCode ()
#15 0x00005555557b87d8 in ()
#16 0x00005555557b20bb in ()
#17 0x00005555557b8525 in ()
#18 0x00005555557b7a08 in _PyRun_SimpleFileObject ()
#19 0x00005555557b7653 in _PyRun_AnyFileObject ()
#20 0x00005555557aa41e in Py_RunMain ()
#21 0x0000555555780cad in Py_BytesMain ()
#22 0x00007ffff7c29d90 in __libc_start_call_main (main=main@entry=0x555555780c70, argc=argc@entry=6, argv=argv@entry=0x7fffffffdfb8) at ../sysdeps/nptl/libc_start_call_main.h:58
#23 0x00007ffff7c29e40 in __libc_start_main_impl
(main=0x555555780c70, argc=6, argv=0x7fffffffdfb8, init=, fini=, rtld_fini=, stack_end=0x7fffffffdfa8) at ../csu/libc-start.c:392
#24 0x0000555555780ba5 in _start ()
==============================================================

scpoppin1207, 2024-10-26:
Exactly the same problem! I'm about to crazy since I have fight with issue for 2 days!

> How to solve this problem??? I also have this problem. I found that the official code of fase_gicp used the main branch. But when I run the command "bash tum.sh", it was not well running. It endless loop(it seems that) at the second iter(current_pose = self.reg.align(initial_pose)). The console has no log output. When I change the fase_gicp brance of gs_gicp_slam, the error changed as : AttributeError: 'pygicp.FastGICP' object has no attribute 'set_max_knn_distance'.
> 
> What should I do???
> 
> ## When I use the gdb debug, I found the **endless loop** is this log:
> ## Thread 1 "python" received signal SIGSEGV, Segmentation fault.
> 0x00007fffc3b0221c in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > > >::calculate_source_covariances_with_filterpcl::PointXYZ(pcl::PointCloudpcl::PointXYZ::ConstPtr const&, pcl::search::Searchpcl::PointXYZ&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator >&, std::vector<float, std::allocator >&, std::vector<int, std::allocator >&) [clone ._omp_fn.0] () from /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
> (gdb) backtrace
> #0 0x00007fffc3b0221c in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > > >::calculate_source_covariances_with_filterpcl::PointXYZ(pcl::PointCloudpcl::PointXYZ::ConstPtr const&, pcl::search::Searchpcl::PointXYZ&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator >&, std::vector<float, std::allocator >&, std::vector<int, std::allocator >&) [clone ._omp_fn.0] ()
> at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
> #1 0x00007fffc39ae686 in GOMP_parallel () at /lib/x86_64-linux-gnu/libgomp.so.1
> #2 0x00007fffc3b0f226 in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > > >::calculate_source_covariances_with_filterpcl::PointXYZ(pcl::PointCloudpcl::PointXYZ::ConstPtr const&, pcl::search::Searchpcl::PointXYZ&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator >&, std::vector<float, std::allocator >&, std::vector<int, std::allocator >&) ()
> at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
> #3 0x00007fffc3b0f4c1 in fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple > > >::computeTransformation(pcl::PointCloudpcl::PointXYZ&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&) ()
> at /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp/build/lib.linux-x86_64-cpython-310/libfast_gicp.so
> #4 0x00007fffc3b5b25e in () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
> #5 0x00007fffc3b5bcef in () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
> #6 0x00007fffc3b4f2ac in () at /home/gvlib_ljh/.local/lib/python3.10/site-packages/pygicp-0.0.1-py3.10-linux-x86_64.egg/pygicp.cpython-310-x86_64-linux-gnu.so
> #7 0x00005555556b3e0e in ()
> #8 0x00005555556aa5eb in _PyObject_MakeTpCall ()
> #9 0x00005555556c27bb in ()
> #10 0x00005555556a28a2 in _PyEval_EvalFrameDefault ()
> #11 0x00005555556b470c in _PyFunction_Vectorcall ()
> #12 0x000055555569ce0d in _PyEval_EvalFrameDefault ()
> #13 0x000055555578de56 in ()
> #14 0x000055555578dcf6 in PyEval_EvalCode ()
> #15 0x00005555557b87d8 in ()
> #16 0x00005555557b20bb in ()
> #17 0x00005555557b8525 in ()
> #18 0x00005555557b7a08 in _PyRun_SimpleFileObject ()
> #19 0x00005555557b7653 in _PyRun_AnyFileObject ()
> #20 0x00005555557aa41e in Py_RunMain ()
> #21 0x0000555555780cad in Py_BytesMain ()
> #22 0x00007ffff7c29d90 in __libc_start_call_main (main=main@entry=0x555555780c70, argc=argc@entry=6, argv=argv@entry=0x7fffffffdfb8) at ../sysdeps/nptl/libc_start_call_main.h:58
> #23 0x00007ffff7c29e40 in __libc_start_main_impl
> (main=0x555555780c70, argc=6, argv=0x7fffffffdfb8, init=, fini=, rtld_fini=, stack_end=0x7fffffffdfa8) at ../csu/libc-start.c:392
> #24 0x0000555555780ba5 in _start ()

eainx, 2024-12-19:
Did you address this problem?

jxr233, 2025-01-05:
@eainx @scpoppin1207 Not sure if this is still needed, but I might have found the cause of the issue. You can check the version of the PCL library you are using. I experienced the same error with PCL 1.12, but switching to PCL 1.10 solved the problem for me. Hope this helps!

eainx, 2025-01-06:
> @eainx @scpoppin1207 Not sure if this is still needed, but I might have found the cause of the issue. You can check the version of the PCL library you are using. I experienced the same error with PCL 1.12, but switching to PCL 1.10 solved the problem for me. Hope this helps!

Thanks it works perfectly! :D