Lab-of-AI-and-Robotics/GS_ICP_SLAM #27: Questions about visualizing results

Tokyo-dream, 2024-05-16:
Thank you so much for the excellent work!!!
However, the result I got was not like what you showed. My Gaussian was viewed through SuperSplat(https://playcanvas.com/supersplat/editor), which is very messy. There was no problem with the rendered image, PSNR=35.
Here is the rendered image
[image]
Here is the SuperSplat Viewer
[image]

small-zeng, 2024-05-22:
> Thank you so much for the excellent work!!! However, the result I got was not like what you showed. My Gaussian was viewed through SuperSplat(https://playcanvas.com/supersplat/editor), which is very messy. There was no problem with the rendered image, PSNR=35. Here is the rendered image ![result_599](https://private-user-images.githubusercontent.com/63401220/331075492-ebb03fa7-9245-48d0-906f-9a238a1d0987.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzNDA2NzgsIm5iZiI6MTcxNjM0MDM3OCwicGF0aCI6Ii82MzQwMTIyMC8zMzEwNzU0OTItZWJiMDNmYTctOTI0NS00OGQwLTkwNmYtOWEyMzhhMWQwOTg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIyVDAxMTI1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ2MmI5N2RiYjVlYjA0NGFlYjBmNjMwZDg3ZmI4MWY2NzMxMzQ3ZDM4YmMwZGFhMGRlOTVmOTYxNTgwOTlhNzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4pxHHjT7r6OiOJVRG0jH5hkYjavuQWm7VvYAdyv-Fbw) Here is the SuperSplat Viewer <img alt="11" width="1162" src="https://private-user-images.githubusercontent.com/63401220/331076046-ea6b599d-24c9-4850-99b7-fe936babd9cd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzNDA2NzgsIm5iZiI6MTcxNjM0MDM3OCwicGF0aCI6Ii82MzQwMTIyMC8zMzEwNzYwNDYtZWE2YjU5OWQtMjRjOS00ODUwLTk5YjctZmU5MzZiYWJkOWNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIyVDAxMTI1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZhOWU0NzUyY2QxNTlkNDczMDU0NmI4YjE0YmNlMjExYzAxNzhiNDFiMWVjOGU4YzkzZmI3NzZjNmQ3ODc5ZWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NsGpwpdDBIRvoL9Ndx6HFi19e9pwbhLycd6tu9N4KDg">

I have met the same problem, do you have solved it?

Tokyo-dream, 2024-05-22:
> I have met the same problem, do you have solved it?

nope

tarquai, 2024-08-21:
I have used gsplat (look for the python package) and got the same problems.
Investigating a bit i get that the "scale" of the gaussians is very uneven: the z components is much smaller then the rest.

pleituer, 2024-09-08:
I looked into the source code of the rasterization package and it seems that they modified it so the quaternions for the rotation matrix are in the form (x, y, z, w) while the implementation of the original paper and many others uses (w, x, y, z). Swapping the order seems to solve the issue.

GS-ICP-SLAM's implementation in `submodules/diff-gaussian-rasterization/cuda_rasterizer/forward.cu` line 134-145
```c
	glm::vec4 q = rot;// / glm::length(rot);
	float x = q[0];
	float y = q[1];
	float z = q[2];
	float w = q[3];

	// Compute rotation matrix from quaternion
	glm::mat3 R = glm::mat3(
		1.f - 2.f * (y * y + z * z), 2.f * (x * y - w * z), 2.f * (x * z + w * y),
		2.f * (x * y + w * z), 1.f - 2.f * (x * x + z * z), 2.f * (y * z - w * x),
		2.f * (x * z - w * y), 2.f * (y * z + w * x), 1.f - 2.f * (x * x + y * y)
	);
```

Original 3DGS paper's implementation in `submodules/diff-gaussian-rasterization/cuda_rasterizer/forward.cu` line 127-138
```c
	glm::vec4 q = rot;// / glm::length(rot);
	float r = q.x;
	float x = q.y;
	float y = q.z;
	float z = q.w;

	// Compute rotation matrix from quaternion
	glm::mat3 R = glm::mat3(
		1.f - 2.f * (y * y + z * z), 2.f * (x * y - r * z), 2.f * (x * z + r * y),
		2.f * (x * y + r * z), 1.f - 2.f * (x * x + z * z), 2.f * (y * z - r * x),
		2.f * (x * z - r * y), 2.f * (y * z + r * x), 1.f - 2.f * (x * x + y * y)
	);
```

tarquai, 2024-09-08:
you changed the "forward", now you need to change the "backword"?
how does this change effect the results in the paper?

pleituer, 2024-09-08:
I didn't make modifications to the existing code, instead I just created a new script that loads the ply file and corrects the rotation so it matches with the implementation of the original paper. And I didn't find any changes to the results as both are just different representations of the same rotation matrix.

tarquai, 2024-09-09:
Thanks you.
The visualization looks much better now.
The simplest thing to do is write:
    quats = quats[:, [3, 0, 1, 2]]
just  before your "visualization function" of your choice.
And there is no other change needed other then this?

tarquai, 2024-09-10:
why is the "scale" of the gaussians so uneven? (z is much smaller then the rest)?  I thought this is the cause of the issue