Lab-of-AI-and-Robotics/GS_ICP_SLAM #36: Custom Dataset

ddkats, 2024-07-05:
Hi @Riboha 

Thank you for uploading this great work. I have a couple of questions:

1) How can I test the code with my own dataset?
2) Is it necessary to have the trajectory.txt file if we don't care about the evaluation results? Can we use the pipeline with just pairs of RGB and depth frames?

BR

Riboha, 2024-07-06:
Hello,

1. It is not included in this repo. Adding GS-ICP SLAM capabilities to work with user's own datasets is on my plan, but I'm uncertain when this update will be released. I'm very sorry about this.
2. The ground truth trajectory.txt is only used for evaluating results. Therefore, if evaluation is not required, the ground truth trajectory is unnecessary.

ddkats, 2024-07-06:
> Hello,
>
> 1. It is not included in this repo. Adding GS-ICP SLAM capabilities to work with user's own datasets is on my plan, but I'm uncertain when this update will be released. I'm very sorry about [...]
Thank you for your quick response. Just to clarify, will it not work if I provide the data in this format:

    - scene
        - images (contain rgb images)
            - frame000000.jpg
            ...
        - depth_images (contain depth images)
            - depth000000.jpg

Saligz, 2024-07-18:
Amazing work ! I also would be interested in a bit more info to use with my own dataset