Lab-of-AI-and-Robotics/GS_ICP_SLAM #37: gs_gicp_slam did not work for GS_ICP_SLAM

huarzail, 2024-07-17:
Hello, all authors. 

At first, thank you for your contribution to the open source community.

When I installed the gs_gicp_slam, I run the command"bash tum.sh",it seems that this project endless loop at the code "current_pose = reg.align(poses[-1])" in mp_Tracker.py. Some scholar tell me that may be the fast_gicp was not installed correctly.

When I installed the fast_gicp again, I run the example in the python_tester directory(python using_previous_30.py ../../../../datasets/TUM/rgbd_dataset_freiburg3_long_office_household tum 0.05 false). It breaked, the log is "Segmentation fault (core dumped)". Also caused by the code "current_pose = reg.align(poses[-1])".

Then I use the gdb debug, the log is:

Thread 43 "python" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff48f62640 (LWP 2740189)]
0x00007fffc3b0221c in bool fast_gicp::FastGICP<pcl::PointXYZ, pcl::PointXYZ, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >, pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > > >::calculate_source_covariances_with_filter<pcl::PointXYZ>(pcl::PointCloud<pcl::PointXYZ>::ConstPtr const&, pcl::search::Search<pcl::PointXYZ>&, std::vector<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >&, std::vector<float, std::allocator<float> >&, std::vector<float, std::allocator<float> >&, std::vector<int, std::allocator<int> >&) [clone ._omp_fn.0] () from /home/gvlib_ljh/3DGS-SLAM/GS_ICP_SLAM/submodules/fast_gicp_slam/build/lib.linux-x86_64-cpython-310/libfast_gicp.so

Will you can tell me how to fix this error.

Thanks.

Aaron-0303, 2024-11-02:
I also encountered the same problem. May I ask if it has been resolved?

Justzhongjj, 2024-11-20:
I also encountered the same problem, too. May I ask if it has been resolved?

Liu-Sdu, 2025-02-15:
I also encountered the same problem, too. May I ask if it has been resolved?