Lab-of-AI-and-Robotics/GS_ICP_SLAM #44: Do not have a good result on TUM

Zhaoyibinn, 2024-09-10:
Hello, thanks you for your open access,but I have met some problem during running.
When I use the replica dataset, there are basically no problems running, but when I use the TUM dataset, my results are significantly lower than those in the paper and are basically unusable:
PSNR: 15.88
ATE RMSE : 21.90
s
[image]
Is there any problem??

Lucasmogsan, 2024-10-10:
I have similar problem with even worse RMSE (>20) just running it as is on the TUM freiburg1_desk:

`python -W ignore gs_icp_slam.py --dataset_path /home/GS_ICP_SLAM/dataset/TUM/rgbd_dataset_freiburg1_desk --config /home/GS_ICP_SLAM/configs/TUM/rgbd_dataset_freiburg1_desk.txt --rerun_viewer`

I achieve the following:

```
System FPS: 29.94
ATE RMSE: 33.15
PSNR: 13.92
SSIM: 0.635
LPIPS: 0.434
```
... which is not good...
I use the docker-container on ubuntu and have a RTX 3060 Graphics card.

@bogus2000 @Riboha : Is this because some parameters are specifically tuned to particular datasets?

nico0704, 2024-11-05:
Have you tried running it with the same parameters as provided in tum.sh?

overlapped_th=1e-3
max_correspondence_distance=0.03
knn_maxd=99999.0
trackable_opacity_th=0.09
overlapped_th2=1e-3
downsample_rate=5
keyframe_th=0.81

```bash
python -W ignore gs_icp_slam.py --dataset_path /home/GS_ICP_SLAM/dataset/TUM/rgbd_dataset_freiburg1_desk --config /home/GS_ICP_SLAM/configs/TUM/rgbd_dataset_freiburg1_desk.txt --overlapped_th=1e-3 --max_correspondence_distance=0.03 --knn_maxd=99999.0 --trackable_opacity_th=0.09 --overlapped_th2=1e-3 --downsample_rate=5 --keyframe_th=0.81 --rerun_viewer
```

maybe even try a higher value for the max_correspondence distance like 0.1

This worked for us. Replica might behave quite different in comparison to real-world datasets due to its synthetic nature, hence the parameters should differ as well to get good results.

Lucasmogsan, 2024-11-05:
@nico0704 Thanks for your reply!

We did get some better results with the parameters from the tum.sh script, thank you!
Will look more into this method - have you been trying it on other or your own datasets?

nico0704, 2024-11-05:
@Lucasmogsan 

We tried it with both TUM and our own dataset and got some alright looking results.
You can check out our [fork](https://github.com/nico0704/GS_ICP_SLAM) of the repo which includes adaptions for using custom datasets as well as some basic scripts to run a live version of the algorithm where images get "streamed" from a realsense camera and then processed by the algorithm.
We included some visualizations of our own data as well if you're interested :)

https://github.com/nico0704/GS_ICP_SLAM

Lucasmogsan, 2024-11-05:
@nico0704 amazing work and thanks for sharing. I think we can make good use of your contribution indeed.

Aoko955, 2025-01-01:
> @Lucasmogsan
>
> We tried it with both TUM and our own dataset and got some alright looking results. You can check out our [fork](https://github.com/nico0704/GS_ICP_SLAM) of the repo which includes [...]

Thank you for your good work. I am trying to use your code to run my dataset. But an error was reported with the following message：
[pcl::FastGICP::setInputTarget] Invalid or empty point cloud dataset given!
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[pcl::FastGICP::setInputSource] Invalid or empty point cloud dataset given!
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!

Is this because my dataset is bad? Can you upload your dataset to your repository?

Aoko955, 2025-01-01:
> We tried it with both TUM and our own dataset and got some alright looking results. You can check out our [fork](https://github.com/nico0704/GS_ICP_SLAM) of the repo which includes adaptions for [...]

@nico0704
Thank you for your good work. I am trying to use your code to run my dataset. But an error was reported with the following message：
[pcl::FastGICP::setInputTarget] Invalid or empty point cloud dataset given!
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[pcl::FastGICP::setInputSource] Invalid or empty point cloud dataset given!
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!

Is this because my dataset is bad? Can you upload your dataset to your repository?

Aoko955, 2025-01-04:
@Lucasmogsan 
Have you tried it on your own dataset?  I am trying to run it on my dataset. But an error was reported with the following message：
[pcl::FastGICP::setInputTarget] Invalid or empty point cloud dataset given!
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[pcl::FastGICP::setInputSource] Invalid or empty point cloud dataset given!
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!